nvs-vocabs/L22 #207: NTR: Eppley Standard Precision Pyranometer

rhudak2020, 2023-11-22:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Eppley

## Device range
n/a

## Device model
SPP

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
pyranometer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
Eppley SPP

## Documentation
http://www.eppleylab.com/instrument-list/standard-precision-pyranometer/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2024-02-22:
This term has now been loaded and will be live on the NVS from tomorrow morning:

TOOL2020 - Eppley Standard Precision Pyranometer

danibodc, 2024-02-23:
http://vocab.nerc.ac.uk/collection/L22/current/TOOL2020/

closed 2024-02-23 by danibodc