nvs-vocabs/L22 #210: NTR: Light trap

timvdstap, 2023-11-28:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Unspecified

## Device range
n/a

## Device model
n/a

## Model version
n/a

## Model alternative name (if applicable)
Shanks-style Light Trap

## Modification (if applicable)

Light trap has a large float on top to keep it upright in the water column and a weight at the very bottom to keep a large jug submerged in the water near the surface. Large funnels are attached to the sides of the jug that allow animals to swim in, but not out. At night, a timer inside the trap turns on a strip of LED lights, attracing animals to the trap. When pulling the trap out of the water, the sample drains through a cod-end, which helps filter the water out but collecting the animals.

## Qualification (if applicable)
portable

## Instrument type
Light trap used to attract small animals at night time.

## Does the tool describe a series of instruments? 
n/a

## Reference (if applicable)
https://www.pacshell.org/light-trap.asp
https://www.pacshell.org/pdf/houle_s4_summary_final.pdf

## Short name
Light trap

## Documentation
https://www.pacshell.org/light-trap.asp
https://www.pacshell.org/pdf/houle_s4_summary_final.pdf

timvdstap, 2023-11-28:
@danibodc Sorry! I didn't realize you had changed the title and I added 'LED' to it. I'm not sure if [BODCNVS-1965] still applies?

JordanAthertonNOC, 2023-11-28:
No problem @timvdstap, for your information that is just our way of tracking and allocating requests internally - I have added it back on as it still applies.

timvdstap, 2024-05-15:
Hi @danibodc would you happen to have an update on this NTR? :) Thank you in advance!

roswri, 2024-05-16:
Hi @timvdstap,

I've working on it as we speak, should be done by the end of the week :)

Thanks,
Roseanna

roswri, 2024-05-17:
Hi @timvdstap,
The new term is created and live on the NVS:
Shanks-style LED Light Trap (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2026/)

closed 2024-05-17 by roswri