nvs-vocabs/L22 #211: NTR: C-Nav X1 GNSS

rhudak2020, 2023-12-01:
## Device make
Oceaneering

## Device range
C-nav

## Device model
X1

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
gnss 

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
C-Nav X1

## Documentation
https://www.oceaneering.com/C-Nav/C-NavSupport/C-NavX1/CNAV-MAN-061.0%20%28C-NavX1%20User%20Guide%29.pdf

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2024-05-29:
This term has now been loaded and will be live on the NVS from tomorrow morning:

TOOL2034	C-NavX1 Global Navigation Satellite System (GNSS) receiver

danibodc, 2024-05-30:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2034/

closed 2024-05-30 by danibodc