nvs-vocabs/L22 #215: NTR: Apollo SciTech AS-C6L DIC analyzer

robthomas-marine, 2024-01-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Apollo SciTech

## Device range
DIC analyzer

## Device model
AS-C6L 

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
Apollo SciTech's new AS-C6L DIC analyzer has been developed and tested for total dissolved inorganic carbon analysis in a variety of aquatic environments. Advantages of this DIC analyzer include the use of small sample volumes (0.5 – 3.5 mL) with a fast analytical rate (~ 3 minutes per analysis). Further, the system maintains a precision of ± 0.1% or better.

The AS-C6L DIC analyzer is comprised of a laser-based CO2 detector (LI-7815, LI-COR, USA), a digital syringe pump for precise delivery of reagent and sample solutions (Tecan, Switzerland), a mass flow controller to accurately regulate the carrier gas flow rate, a specifically designed CO2 stripping reactor that allows for an efficient and smooth stripping of CO2 from the water sample, an electronic cooling system for removing moisture, and a computer communication assembly.

## Instrument type
DIC analyzer

## Does the tool describe a series of instruments? 
n/a

## Reference (if applicable)
n/a

## Short name
Apollo SciTech AS-C6L DIC analyzer

## Documentation
https://www.apolloscitech.com/dic-c6l.html

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
n/a

danibodc, 2024-05-29:
This term has now been loaded and will be live on the NVS from tomorrow morining:

TOOL2033	Apollo SciTech AS-C6L Dissolved Organic Carbon (DIC) analyser

danibodc, 2024-05-30:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2033/

closed 2024-05-30 by danibodc