nvs-vocabs/L22 #217: NTR: UWITEC KOL Kolbenlot percussion piston corer

jamcas-bas, 2024-01-17:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
UWITEC

## Device range
n/a

## Device model
Kolbenlot

## Model version
n/a

## Model alternative name (if applicable)
KOL

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
percussion piston corer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
UWITEC piston corer

## Documentation
https://web.archive.org/web/20031128085831/http://www.uwitec.at/html/frame.html (Site Map > Piston corer)
https://www.uib.no/en/geo/112078/uwitec-piston-corer

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

danibodc, 2024-06-07:
This has been loaded to L22 and will be live on the NVS from tomorrow morning:

TOOL2046	UWITEC percussion piston corer

danibodc, 2024-06-10:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2046/

closed 2024-06-10 by danibodc