nvs-vocabs/L22 #224: NTR: Simrad ES200-7C

rhudak2020, 2024-01-23:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Kongsberg

## Device range
Simrad

## Device model
ES200-7C

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
split beam transducer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
ES200-7C

## Documentation
https://www.kongsberg.com/maritime/products/commercial-fisheries/td/200-khz/simrad-es200-7c/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2024-02-22:
This term has now been loaded to L22 and will be live on the NVS from tomorrow morning:

TOOL2019	Kongsberg (Simrad) ES200-7C split-beam echosounder transducer

danibodc, 2024-02-23:
http://vocab.nerc.ac.uk/collection/L22/current/TOOL2019/

closed 2024-02-23 by danibodc