nvs-vocabs/L22 #230: NTR: Amersham Imager 600

whshannon, 2024-02-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
GE

## Device range
n/a

## Device model
Amersham™ Imager

## Model version
600

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
1D/Western blotting & quantitative imaging.
Amersham Imager 600 is a CCD camera system that produces digitial images of samples in gels or membranes in three different modes: chemiluminescence, colorimetric, and fluorescence

## Does the tool describe a series of instruments? 
n/a

## Reference (if applicable)
n/a

## Short name
AI600

## Documentation
https://pdf.medicalexpo.com/pdf/ge-healthcare-life-sciences/amersham-imager-600/80554-135352.html
https://lab.vanderbilt.edu/cdb-core-lab/ai600-chemiluminescent-imager/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-9081-1004

jakjonbodc, 2024-06-28:
Good morning,

This L22 term is now available on NVS.

https://vocab.nerc.ac.uk/collection/L22/current/TOOL2058/

Many thanks,
Jak

closed 2024-06-28 by jakjonbodc

whshannon, 2024-07-01:
Thanks @jakjonbodc !