nvs-vocabs/L22 #233: NTR:HCMR Hard Substrate Sampler

melinalou, 2024-03-19:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Hellenic Centre for Marine Research

## Device range
n/a

## Device model
Hard Substrate Sampler

## Model version
n/a

## Model alternative name (if applicable)
HCMR Hard Substrate Sampler

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
Sampler

## Does the tool describe a series of instruments? 
Individual instrument

## Reference (if applicable)
Not provided

## Short name
HCMR HSS

## Documentation
Hellenic Centre for Marine Research [custom] hard substrate sampler -
Scuba diving, by placing a plexiglas frame (25x25 cm) with attached net 63μm on the substrate and scraping the framed area completely. The samples were then collected with an underwater suction device working with pressure from a diving tank. Samples were washed through a 500 mm mesh sieve and fixed in 96% ethanol

## Mapping to external terminologies (optional)
Not provided

## ORCID (optional)
Not provided

melinalou, 2024-04-23:
Hi, is it possible to know about when it will be published?
thank you!

danibodc, 2024-04-23:
Hi @melinalou we are working through our backlog, do you have an urgent requirement for this term? If so do let us know so we can prioritise.

Thank you!

melinalou, 2024-04-24:
Hi @danibodc and thank you very much for the respond!
Yes if possible please prioritize because we need to publish the dataset.
Have a nice day!

danibodc, 2024-04-24:
Hi @melinalou, is this device different to [TOOL1815](https://vocab.nerc.ac.uk/collection/L22/current/TOOL1815/)?

gwemon, 2024-04-29:
Hi @danibodc @melinalou if this is the case the n should we not consider broadening the defintion of TOOL1815? What are these minor differences? Are they iimprovements? or adaptation for use in different conditions?

melinalou, 2024-04-29:
Hi @gwemon Yes they are adaptation for use in different conditions.

vpaba, 2024-05-07:
Dear @melinalou,

Thanks for your patience. I've been looking into this and I have two questions: 

1. Could the plexiglass frame be used independently of the MANOSS, and could the MANOSS be used independently of the plexiglass frame? 
2. If the MANOSS has to be used in conjunction with a frame, does it have to be made of plexiglass, and have the size, shape and mesh size specified above, or can variations be used as well?

Thank you and kind regards,
Violetta

JordanAthertonNOC, 2024-07-24:
Hi @melinalou 

I have taken over management of this request, is there any update on Violetta's questions from above? 

If not we may look at closing this request. 

Best

Jordan

melinalou, 2024-07-24:
Hi @JordanAthertonNOC and sorry for the delay!
Unfortunately I don't have an answer so far, so we maybe should close the issue.

All the best
Melina

JordanAthertonNOC, 2024-07-25:
Thanks @melinalou, feel free to reopen this at a later date if you want to revive this issue.

closed 2024-07-25 by JordanAthertonNOC

melinalou, 2024-12-09:
Hi again! I 've made some new requests in this form https://docs.google.com/forms/d/e/1FAIpQLSe-ZTKJZmNT5FmyPRFsPsNt2hpB_gb6MAmm7Zp-7GSMpn5NFA/viewform and I would like to ask you if it is under your responsibility. If yes, is there a way to know when the new parameters will be on the nvs vocab?
Thank you very much,
Melina

danibodc, 2024-12-09:
Hi @melinalou, yes this form is a way of submitting issues to nvs-vocabs and will place them in the repository associated with the collection the requests are related to. In this case as they were requests for new P02 terms, they can be found at https://github.com/nvs-vocabs/P02/issues. We will be in contact with you soon about some of these as P01 might be more appropriate for some of these, and some of your requests might not be suitable for P02.

melinalou, 2024-12-10:
Thank you very much!