nvs-vocabs/L22 #236: NTR: Forssea Robotics SMOOVE V1 (4K) video camera

bhrtd, 2024-04-19:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Forssea Robotics

## Device range
N/A

## Device model
SMOOVE

## Model version
V1

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
Resident Subsea 4k

## Instrument type
Video Camera

## Does the tool describe a series of instruments? 
System

## Reference (if applicable)
https://oceanroboticsplanet.com/forssea-robotics-delivers-two-smoove-resident-subsea-vision-systems-to-ifremer-03-07-2023

## Short name
FORSSEA ROBOTICS SMOOVE V1

## Documentation

- Manufacturer website and datasheet: https://forssea-robotics.fr/smart-cameras

- Instrument Manual (not available online): "Forssea Robotics SMOOVE is a streaming video camera integrated into the Tempo-Mini module. SMOOVE is a resident Camera System designed for real-time underwater vision and scientific observatories deployment, featuring the recording and streaming of 4K photos and videos during predetermined cycles. This camera system includes a low-energy chlorination system to prevent marine growth on viewports and lights, allowing for extended use in deep-sea applications. The integration of a low-energy chlorination system in front of the camera viewports mitigates the risk of biofouling. It is equipped with 2 LED lights and an anti-biofouling system for optical sensor parts. The system offers a turnkey API, a web user interface, and multiple customization options (depth, lights, cables & power supply). It also features a low-power consumption chlorination system in front of the camera and lights."

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
N/A

bhrtd, 2024-05-17:
Hi, I wanted to follow up on the status of this issue

danibodc, 2024-06-06:
This term has now been loaded to L22 and will be live on the NVS from tomorrow morning:

TOOL2045	Forssea Robotics SMOOVE Resident Camera System

bhrtd, 2024-06-06:
Thanks for the update!

danibodc, 2024-06-07:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2045/

closed 2024-06-07 by danibodc