nvs-vocabs/L22 #238: NTR: Thermo Scientific Partisol 2025i Sequential Air Sampler

whshannon, 2024-04-22:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Thermo Scientific

## Device range
Sequential Air Sampler

## Device model
Partisol 2025i

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
Sequential Air Sampler

## Instrument type
Particulate Detection Monitor

## Does the tool describe a series of instruments? 
n/a

## Reference (if applicable)

## Short name
Partisol 2025I

## Documentation
https://www.thermofisher.com/order/catalog/product/2025I

## Mapping to external terminologies (optional)

## ORCID (optional)
0000-0002-9081-1004

roswri, 2024-06-17:
Hello @whshannon ,
This term has now been created and should be live on the NVS early tomorrow: Thermo Scientific Partisol 2025i sequential air sampler (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2048/)

closed 2024-06-17 by roswri

whshannon, 2024-07-01:
Thank you @roswri!