nvs-vocabs/L22 #239: NTR: Sunburst AFT-CO2

jnahorniak, 2024-04-30:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Sunburst Sensors

## Device range
n/a

## Device model
AFT-CO2

## Model version
n/a

## Model alternative name (if applicable)
Autonomous Flow Through Instrument carbon dioxide

## Modification (if applicable)

## Qualification (if applicable)
autonomous

## Instrument type
dissolved gas sensor

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)

## Short name
Sunburst AFT-CO2

## Documentation
http://www.sunburstsensors.com/products/oceanographic-carbon-dioxide-sensor-benchtop.html

## Mapping to external terminologies (optional)
http://vocab.nerc.ac.uk/collection/B75/current/ORG00366/
http://vocab.nerc.ac.uk/collection/L05/current/351/

## ORCID (optional)

JordanAthertonNOC, 2024-07-11:
Hi Jasmine

Thank you for your request, I have now created a new L22 term for this instrument, the link will not resolve until ~23:00 PDT. 
Sunburst Autonomous Flow Through Instrument {AFT-CO2} pCO2 sensor (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2067/)

closed 2024-07-11 by JordanAthertonNOC

jnahorniak, 2024-07-12:
> [This email originated from outside of OSU. Use caution with links and attachments.]

Thank you so much!

Jas :)