nvs-vocabs/L22 #243: NTR: Markey COM-7E Electric Single Drum Research Winch

snionbaven, 2024-05-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Markey

## Device range
COM

## Device model
7E

## Model version
v18414

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'compact','lightweight','variable-speed','multipurpose'

## Instrument type
Compact, lightweight, single drum research winch with variable speed electric motor drive and level-wind designed for use with industry standard CTD's, water sampling carousels, and payloads typically used on inshore and coastal research vessels.

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

## Short name
n/a

## Documentation
(https://pdf.nauticexpo.com/pdf/markey-machinerympany/com-7e-v18414/28204-38001.html)

## Mapping to external terminologies (optional)

## ORCID (optional)

danibodc, 2024-05-10:
Hi @snionbaven, many thanks for your L22 requests! Do you have any urgency or deadline for these to be completed?

snionbaven, 2024-05-10:
Good afternoon, @danibodc . No urgency or deadline, thank you for reaching out! Have a great weekend!

gwemon, 2024-05-13:
Hello @snionbaven which organisation and or which project are these codes for please? Many thanks. Gwen.

rhudak2020, 2024-05-13:
Hi @gwemon these are for the R2R project

jakjonbodc, 2024-07-18:
Hello @snionbaven,

Thank you for submitting a request for a new term. Unfortunately this term does not meet the conditions of our L22 vocabulary, which is designed to contain instruments and sensors that are used to produce data, nor our B76 vocabulary for platforms.

Please do let me know if you have any questions on this matter.

Kind regards,
Jak Jones

closed 2024-07-25 by JordanAthertonNOC