nvs-vocabs/L22 #247: NTR: Kipp & Zonen CGR4 Pyrgeometer

snionbaven, 2024-05-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Kipp & Zonen

## Device range
n/a

## Device model
CGR4

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'180-field-of-view','negligible-directional-response-error','hard-carbon-coating'

## Instrument type
Pyrgeometer

## Does the tool describe a series of instruments? 
Instrument

## Reference (if applicable)
n/a

## Short name
n/a

## Documentation
https://www.kippzonen.com/Product/17/CGR4-Pyrgeometer#.XpSK91P7Q6U

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)

JordanAthertonNOC, 2024-07-31:
Hi @snionbaven, this instrument already exists in the L22 collection:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1333/

snehalprabhu10, 2024-07-31:
Hi @JordanAthertonNOC 

Here is the link to the instrument:

JordanAthertonNOC, 2024-08-13:
Hi @snionbaven, @snehalprabhu10 

One new term has been created and will be published online tomorrow as follows:
Kipp and Zonen CG4 pyrgeometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2083/)

closed 2024-08-13 by JordanAthertonNOC