nvs-vocabs/L22 #248: NTR: Kipp & Zonen CMP21 Pyranometer

snionbaven, 2024-05-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Kipp & Zonen 

## Device range
n/a

## Device model
CMP21

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'verified-cosine-response','verified-temperature-dependence','low-dome-IR-offset-error'

## Instrument type
Pyranometer

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
n/a

## Short name
n/a

## Documentation
https://www.kippzonen.com/Product/14/CMP21-Pyranometer#.XpSPE1P7Q6U

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)

JordanAthertonNOC, 2024-07-31:
Hi @snionbaven, this instrument already exists in the L22 collection:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1502/

snehalprabhu10, 2024-07-31:
Hi @JordanAthertonNOC 

The original request was for the instrument: Kipp & Zonen CM21 Pyranometer. Would it be possible to create a new vocab term for that? 

Here is the link to the instrument manual: https://www.kippzonen.com/Download/52/CM-21-Pyranometer-Manual

JordanAthertonNOC, 2024-08-13:
Hi @snionbaven, @snehalprabhu10 

One new term has been created and will be published online tomorrow as follows:
Kipp and Zonen CM21 pyranometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2084/)

closed 2024-08-13 by JordanAthertonNOC