nvs-vocabs/L22 #250: NTR: Gems RFO-2500 Flow Sensor

snionbaven, 2024-05-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Gems

## Device range
n/a

## Device model
RFO-2500

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'continuous-flow'

## Instrument type
Electronic Flow Sensor

## Does the tool describe a series of instruments? 
Individual 

## Reference (if applicable)
n/a

## Short name
n/a

## Documentation
n/a

## Mapping to external terminologies (optional)

## ORCID (optional)

JordanAthertonNOC, 2024-07-31:
Hi @snionbaven 

We created a term covering the RFO/RFA-2500 series in a previous request here:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2071/

Does this fit for you or would you prefer a specific term for the RFO-2500?

JordanAthertonNOC, 2024-08-13:
Closing this ticket, feel free to reopen if this specific model term is required.

closed 2024-08-13 by JordanAthertonNOC