nvs-vocabs/L22 #258: NTR: Kongsberg EM2040 Multibeam Echosounder

snionbaven, 2024-05-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Kongsberg

## Device range
EM2040 

## Device model
EM2040C 

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'shallow-water'

## Instrument type
Multibeam Echosounder

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
n/a

## Short name
Kongsberg EM2040C

## Documentation
https://www.kongsberg.com/discovery/seafloor-mapping/EM2040C-MkII/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)

rhudak2020, 2024-05-14:
Hi @danibodc you can go ahead and remove this one, I was able to find this term already in the NERC database (TOOL1528). Thanks and sorry for any confusion!

danibodc, 2024-05-14:
Hi @rhudak2020 

I think the EM2040C is actually slightly different than the original EM2040 (TOOL1528) - it is a shallow-water system depth-rated to 50 m, also available as a EM2040CX, depth-rated to 1500 m (https://www.kongsberg.com/globalassets/maritime/km-products/product-documents/369468_em2040c_product_specification.pdf).

An MKII version of the EM2040C was released later, which is depth-rated to ~500 m (https://www.kongsberg.com/discovery/seafloor-mapping/EM2040C-MkII/). The original EM2040 is rated to ~6000 m. 

So I think it would still be worth us creating one or more new terms here for the various versions of the EM2040C. Do you happen to know which you would need (i.e. the EM2040C, EM2040CX, or EM2040C MKII)?

rhudak2020, 2024-05-14:
Hi @danibodc,
I looked into our documentation a little more and we have it recorded for the Falkor (too). In there website they refer to it as a  'EM2040 MKII', but go on to say that they use it for 'to about 400 meters' which makes it sound more like the EM2040C version.

https://schmidtocean.org/sonars-acoustics-and-mapping/

ahpanton, 2024-09-05:
Hi @rhudak2020 

I will be creating this term but just wanted to double check which one is needed - the EM2040C MK II or the EM2040P? Or would it be best to create both if that would be more useful for you?

rhudak2020, 2024-09-05:
Please create a EM2040C MK II term. Thank you!

ahpanton, 2024-09-05:
OK I will get going with that. Thank you!

ahpanton, 2024-09-06:
New term has been created and will be live on NVS from tomorrow (07/09/2024):

TOOL2095 - Kongsberg EM 2040C MKII multibeam echosounder (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2095)

Please let me know if there are any issues
Thanks

closed 2024-09-09 by ahpanton