nvs-vocabs/L22 #259: NTR: Tekron NTS 03-G+

rhudak2020, 2024-05-09:
## Device make
Tekron

## Device range
n/a

## Device model
NTS

## Model version
03-G+

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
time server

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
NTS 03-G+

## Documentation
https://chronos.uk/product/power-master-clocks-timing-distribution/multiport-network-time-server-nts-03-g/#:~:text=Meeting%20the%20demanding%20precision%20and,IRIG%2DB%20option%20delivers%20legacy

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

ahpanton, 2024-09-05:
New L22 created:

TOOL2093 - Tekron NTS 03-G+ network time server

ahpanton, 2024-09-06:
[TOOL2093](http://vocab.nerc.ac.uk/collection/L22/current/TOOL2093/) - Tekron NTS 03-G+ network time server

closed 2024-09-06 by ahpanton