nvs-vocabs/L22 #262: NTR: Vaisala PTU330TS Pressure, Humidity, and Temperature Transmitter

snionbaven, 2024-05-11:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Vaisala

## Device range
PTU300

## Device model
PTU330TS

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
'barometer','humidity','temperature','combined-unit'

## Instrument type
Meteorological Measurement System

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
n/a

## Short name
Vaisala PTU330TS

## Documentation
https://www.vaisala.com/en/products/weather-environmental-sensors/pressure-humidity-temperature-transmitter-ptu300

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)

danibodc, 2024-08-29:
Hi @snionbaven, I just wanted to clarify the exact model you are requesting here.

You have mentioned both the PTU300 and the PTU330TS, but have provided a link to the PTU300 documentation. We already have an entry in L22 for the PTU300: https://vocab.nerc.ac.uk/collection/L22/current/TOOL0741/. 

I haven't managed to find any reference to a PTU330TS, but I have found a PT**B**330TS Barometric Pressure Transfer Standard at https://www.vaisala.com/en/products/devices/instruments/ptb330ts - is this what you are referring to?

snionbaven, 2024-08-29:
Good morning, @danibodc ,
The PTU300 entry for the series of models that already exists works just fine, the pressure transfer standard is not the device being referenced. Thank you!

danibodc, 2024-08-29:
Hi @snionbaven ok that's great, thank you! Will close the ticket.

closed 2024-08-29 by danibodc