nvs-vocabs/L22 #263: NTR:

snehalprabhu10, 2024-05-14:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
General Oceanics

## Device range
pCO2

## Device model
8060

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
Dissolved gas sensor

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
GO pCO2 8060

## Documentation
https://www.generaloceanics.com/8060-pco2-measuring-system.html

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)

danibodc, 2024-09-05:
This has now been loaded and will be live on the NVS from tomorrow morning.

TOOL2094	General Oceanics model 8060 pCO2 measuring system

danibodc, 2024-09-06:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2094/

closed 2024-09-06 by danibodc