nvs-vocabs/L22 #265: NTR: MIDAS SVX2

snehalprabhu10, 2024-05-16:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
MIDAS

## Device range
SVX2

## Device model
MIDAS SVX2

## Model version
n/a

## Model alternative name (if applicable)
SVX2

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
CTD profiler

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
n/a

## Short name
n/a

## Documentation
https://www.valeport.co.uk/products/midas-svx2/

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

ahpanton, 2024-09-10:
New term created - this will be live on the NVS from tomorrow (11/09/2024)

TOOL2096 - Valeport MIDAS SVX2 combined SVP/CTD profiler (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2096)

ahpanton, 2024-09-12:
Term is now live:

[TOOL2096](http://vocab.nerc.ac.uk/collection/L22/current/TOOL2096/) - Valeport MIDAS SVX2 combined SVP/CTD profiler

closed 2024-09-12 by ahpanton