nvs-vocabs/L22 #268: NTR: Gill GMX560 / 4H Jena CONTROS / In-situ Aqua TROLL / Valeport MIDAS ECM

robthomas-marine, 2024-05-30:
Hi,

Please could the instruments in the attached template be added to the L22 vocabulary?

[L22_submission_template_v2.2_MIrequest_20240530.xlsx](https://github.com/nvs-vocabs/L22/files/15499341/L22_submission_template_v2.2_MIrequest_20240530.xlsx)

Cheers
Rob

robthomas-marine, 2024-05-30:
Apologies, see updated spreadsheet with a few extra Aqua TROLL versions included.
[L22_submission_template_v2.2_MIrequest_20240530.xlsx](https://github.com/nvs-vocabs/L22/files/15499405/L22_submission_template_v2.2_MIrequest_20240530.xlsx)

jakjonbodc, 2024-09-24:
Following terms are live on NVS:

https://vocab.nerc.ac.uk/collection/L22/current/TOOL2098/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2099/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2100/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2101/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2102/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2103/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2104/

Closing ticket.