nvs-vocabs/L22 #273: Satlantic/Sea-Bird consistency

callumrollo, 2024-07-31:
Inconsistent treatment of the merger of Sea-Bird and Satlantic.

### Current status

Some instruments list Sea-Bird as the manufacturer, but note that the instrument was developed/previously sold by Satlantic e.g. TOOL1117 "Sea-Bird {Satlantic} HyperOCR spectroradiometer series" includes " initially developed by Satlantic, but currently distributed by Sea-Bird Scientific since the companies combined in 2011"

https://vocab.nerc.ac.uk/collection/L22/current/TOOL1117/

Other sensors are just described as Satlantic e.g. TOOL1292 "Satlantic OCR-504 multispectral radiometer"

https://vocab.nerc.ac.uk/collection/L22/current/TOOL1292/

### Issue for user

This is a cause of confusion. We have Sea-bird built OCR-504 sensors. So we use the manufacturer URI for Sea-Bird Scientific from L35 https://vocab.nerc.ac.uk/collection/L35/current/MAN0013/

But there is no vocab entry for a SeaBird OCR504 in L22. Only one for Satlantic

https://vocab.nerc.ac.uk/collection/L22/current/TOOL0625/

### Potential resolutions

- Use vocabs as is: link to manufacturer SeaBird in L35 and sensor Satlantic OCR-504 in L22 - inconsistent, confusing for end users
- Link to manufacturer Satlantic and sensor Satlantic OCR-504  in L22 - incorrect manufacturer
- Add new vocabulary concepts that treat describe the SeaBird OCR-504 as a separate sensor - seems to defeat the point of controlled vocabularies?
- Edit the vocabs - replace all reference to Satlantic with SeaBird {Satlantic}, as has been done for TOOL1117. These companies merged in 2011, but vocab entries for Satlantic instruments have been added as recently as 2019, so there must be a reason not to do this?

@emmerbodc and @danibodcas this is a common glider sensors, it's a blocker for us adopting OG1 atm.

gwemon, 2024-07-31:
@callumrollo only edited your message to correct Dani's github userid :-)

JordanAthertonNOC, 2024-07-31:
I would propose updating the manufacturer prefix of all Satlantic terms to 
`Satlantic {Sea-Bird}`

This is the same approach we have taken for Sea-Bird's 2010 acquisition of WETLabs, now all Sea-Bird/WETLabs terms are prefixed with 
`WET Labs {Sea-Bird WETLabs}`

Note the subtle difference of including WETLabs within the brace is to account for the widespread use of WET Labs with and without a space, thus to aid searchability. 

More thoughts to follow RE the manufacturer linkage, Dani & I are discussing.

JordanAthertonNOC, 2024-07-31:
The L35 mapping is intended as the name of the company at the time the instrument was made. Concerns about the difficulties/imprecision of identifying which instrument models are pre and post-merger/acquisition (technical spec/datasheets are good indicator but sometimes the parent company will rebrand these). 

Internally at BODC we also link to a term intended as the latest company contact, so for all cases mentioned here this is Sea-Bird. 

Additional grey area in the likely occurrence where an instrument model continues to be manufactured and sold post-merger/acquisition, so you will have instances of that model manufactured by Satlantic and instances manufactured by Satlantic {Sea-Bird}.

louatbodc, 2024-07-31:
L22 essentially reflects the brand of the instrument model at the time which is often reflected in the manufacturer name. Some Satlantic sensors were modified by Seabird after the merger between them, hence some are still Satlantic and some are Seabird. However, some instruments did not change - Seabird merely kept selling them - and the addition of a synonym was added where we or our users felt this was the case. Seabird did continue use the brand WetLabs despite the merger. Not sure this is the case anymore.

JordanAthertonNOC, 2024-08-12:
Hi @callumrollo 

We have discussed this in our L22 team meeting. It's a tricky one, and we have many examples of this in L22 where one company has merged into another, so we've tried to keep this historical lineage behind it. Our preferred course of action is as follows:

- Not to arbitrarily attribute both Satlantic and Sea-Bird to a model unless we know the model was manufactured and distributed by both entities
- If any model has continued to be manufactured by Sea-Bird but has different specs to the Satlantic instances, we can create a new term for Sea-Bird only, linked to the Sea-Bird L35

Some actions I have taken in the meantime:

- One Satlantic term (TOOL1470) was incorrectly linked to the Sea-Bird L35 instead of Satlantic, I have corrected this
- I have conducted a review and updated six Satlantic terms to include {Sea-Bird} in the preferred label where I was able to locate evidence (separate spec sheets) for the same model branded as Satlantic and Sea-Bird. This has impacted all three terms you have mentioned in this ticket (update will be reflected online tomorrow). In the first instance this aids discoverability of the terms

--------------
Hopefully this helps to address this issue for you. Some outstanding questions/comments:

1. L22 and L35 are decoupled in OG1.0, so you should be able to attribute an L35 of your choosing, unless that is, you are trying to do this programmatically - pulling in the respective L35 linked to the L22 from the NVS? (Note many 'legacy' L22 terms (created pre-2019) may not have an L35 mapping but we can add them retroactively on request)

2. Are there other terms causing problems for you that could do with a specific review?

callumrollo, 2024-08-13:
Thanks for the thorough response @JordanAthertonNOC. This resolves the issue for our sensors

closed 2024-08-13 by callumrollo