nvs-vocabs/L22 #282: NTR: CVF4

jnahorniak, 2024-08-13:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Sea-Bird Scientific

## Device range
n/a

## Device model
SBE 11plus

## Model version
V2

## Model alternative name (if applicable)
SBE 11+

## Modification (if applicable)

## Qualification (if applicable)

## Instrument type
CTD deck unit

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)

## Short name
Sea-Bird SBE 11plus

## Documentation
https://www.seabird.com/sbe-911plus-ctd/product?id=60761421595
https://www.seabird.com/product.detail-cms.block.jsa?seoTerm=sbe-911plus-ctd&id=60761421595
(near the bottom of this page there is a description of the 11plus)

## Mapping to external terminologies (optional)
http://vocab.nerc.ac.uk/collection/L22/current/TOOL0058/
http://vocab.nerc.ac.uk/collection/L35/current/MAN0013/
http://vocab.nerc.ac.uk/collection/L22/current/TOOL1508/
Note: There are currently entries for the 911+ and the 9+ in L22, but not the 11+.

## ORCID (optional)

JordanAthertonNOC, 2024-11-12:
Hi @jnahorniak apologies for the delay in the team getting around to this. 

Could you help us better understand the use case here? We do have other [examples](https://vocab.nerc.ac.uk/search_nvs/L22/?searchstr=data%20logger&options=identifier,preflabel,altlabel,status_accepted&rbaddfilter=inc&searchstr2=) in the L22 collection of data loggers, typically where the 'instrument' improves the data in some way, but we don't have any SBE deck units, and this example seems to pass the data from the instrument to a computer, or allow water sample control.

jnahorniak, 2024-11-13:
> [This email originated from outside of OSU. Use caution with links and attachments.]

Jordan,

Thank you for getting back to me on this request!

Regarding the SBE 11 plus (and some of the other unusual vocab items I have requested) ...

We’ve developed software (CORIOLIX) that provides real-time access to data and time-series plots from sensors aboard oceanographic research vessels (from ship and shore).  Part of the configuration includes populating the software with a vessel-specific list of sensors and their associated metadata (e.g. serial numbers, location on board, data format, etc.).  Some of the vessels with CORIOLIX use this software for inventory management in addition to the real-time data feature.

We have adopted the L22 device vocabulary for the sensor models (the vocab is provided in CORIOLIX as a drop-down list).  In some cases, vessels have sensors (or other items that provide data) that can’t be found in the L22 list.  I have been forwarding these items to you as we come across them, so you’ll see more requests trickle in over time.

Specifically, regarding the SBE 11 plus ...
The SBE 11 plus deck box merges data from individual sensors on the CTD with GPS and PAR data into a single message.   For data purposes, we treat the deck box as the data source and define it as the parent of all of its associated child sensors (pH, temperature, 9plus CTD, etc.).  To better track all of the components of the CTD, we store metadata about all of the individual sensors and the deck box in CORIOLIX.  We are currently using the L22 term “Sea-Bird SBE 911plus CTD” to describe the deck box but as this isn’t entirely accurate, so we thought we would request the  SBE 11 plus term.  If this term doesn’t fit your vocab requirements we’ll just keep using the SBE 911plus term.

An example CORIOLIX website (for the R/V Endeavor) can be seen here.  It doesn’t have a CTD example unfortunately, but it will give you some idea of the types of sensors being included. The link below takes you to a table with a long list of sensors.  You’ll see their corresponding L22 vocab names in this table.

https://coriolix.ceoas.oregonstate.edu/endeavor/sensor/list/sensor_name/

If there is an alternative vocabulary that would be more appropriate to use for data loggers, fans, winches, and the like, please let me know!

Thanks!
Jas.

JordanAthertonNOC, 2024-11-14:
Hi @jnahorniak 

Thanks for the thorough reply, it's always great to hear how our users are utilising the vocabulary - I've had a look at the CORIOLIX website in the past out of interest. We've discussed this as a team today and we're happy to create this term for you. Will be back soon with a new term!

jnahorniak, 2024-11-14:
Thank you!!!!

Jas

danibodc, 2024-11-21:
Hi @jnahorniak this term has now been loaded, the URI will be live on the NVS from tomorrow morning.

TOOL2122 - Sea-Bird SBE 11plus V2 deck unit (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2122)

closed 2024-11-22 by danibodc