nvs-vocabs/L22 #284: NTR: Movella Xsens MTi-320 AHT Module

bhrtd, 2024-10-07:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Movella Holdings Inc

## Device range
IP51-rated

## Device model
MTi-320

## Model version
V1

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
Due to the obsolescence of the currently used magnetometer component and consequently to ensure the continuity of supply of pin-compatible MTi-320 products, new sensing elements were incorporated in the hardware design of the MTi-320. This results in the availability of hardware version 3 of the MTi-320. V3 is the direct successor of the hardware version 2. Production of hardware version 3 started in June 2023. Find more details at: https://base.movella.com/s/article/Migration-to-Hardware-Version-3-of-the-MTi-320?language=en_US

## Qualification (if applicable)
Active Heading Tracker

## Instrument type
Xsens MTi-320 AHT

## Does the tool describe a series of instruments? 
MTi 300-series

## Reference (if applicable)
https://www.movella.com/hubfs/Downloads/Leaflets/MTi-320.pdf

## Short name
Movella MTi-320 AHT

## Documentation

- Manufacturer website and datasheet: https://www.xsens.com/hubfs/Downloads/Leaflets/MTi-320.pdf
- The MTi-320 is a high-performing Active Heading tracker (AHT) with a small form factor design for deep integration in your application. The Xsens optimized strap-down algorithm (AttitudeEngineTM) performs high-speed, dead-reckoning calculations at 1 kHz, accurately capturing high-frequency motions. Xsens’ industry-leading sensor fusion algorithm provides high accuracy and sensor auto-calibration in this cost-effective module for a wide range of (embedded) applications. The MTi-320 is supported by the MT Software Suite,  which includes MT Manager (GUI for Windows/Linux), SDK, example codes, and drivers for many platforms including ROS Mapping to external terminologies.

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
N/A

bhrtd, 2024-10-31:
Hi, I'm checking in to follow up on the status of this issue- Thanks!

bhrtd, 2024-11-05:
Hi @roswri , I was hoping to get an update on the status of this request. I’d appreciate your response.

JordanAthertonNOC, 2024-11-07:
Hi @bhrtd 

Apologies we haven't got back to you, Roseanna is presently away at a conference. I can try and take a look tomorrow and hopefully get a code created for you by the middle of next week if that works?

bhrtd, 2024-11-07:
Hi Jordan,

Thanks for your email! Sounds great, yes that works for me.

Thanks again,
Bahareh Torabi | Data Steward
Ocean Networks Canada | oceannetworks.ca
University of Victoria Queenswood Campus
#100-2474 Arbutus Road, Victoria, BC V8N 1V8
*A UNIVERSITY OF VICTORIA INITIATIVE*

*CONFIDENTIALITY NOTICE:* This e-mail message, including any attachments,
is for the sole use of the intended recipient(s) and may contain
confidential and privileged information. Any unauthorized review, use,
disclosure or distribution is prohibited. If you are not the intended
recipient, please contact the sender by reply e-mail and destroy all copies
of the original message.

JordanAthertonNOC, 2024-11-11:
One new term has been created and will be published online tomorrow as follows: 

Movella Xsens MTi-320 active heading tracker (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2119/)

closed 2024-11-11 by JordanAthertonNOC

bhrtd, 2024-11-12:
Hi Jordan,

Thank you so much for allocating a new term and for the heads-up!

Cheers,
Bahareh Torabi | Data Steward
Ocean Networks Canada | oceannetworks.ca
University of Victoria Queenswood Campus
#100-2474 Arbutus Road, Victoria, BC V8N 1V8
*A UNIVERSITY OF VICTORIA INITIATIVE*

*CONFIDENTIALITY NOTICE:* This e-mail message, including any attachments,
is for the sole use of the intended recipient(s) and may contain
confidential and privileged information. Any unauthorized review, use,
disclosure or distribution is prohibited. If you are not the intended
recipient, please contact the sender by reply e-mail and destroy all copies
of the original message.