nvs-vocabs/L22 #285: Animal mounted instrument types

gwemon, 2024-10-24:
Hi,
I am working aligning the WOD Instrument list to L22 and there are two terms I am ensure what to use from the existing L22 concepts:

ANIMAL MOUNTED: Satellite Relay Data Logger (SRDL) (SMRU Instrumentation)
and
ANIMAL MOUNTED: TDR (Time-Depth Recorder) Tag (Wildlife Computers)

I have no other information.
Can you advise?
Many thanks,
Gwen.

JordanAthertonNOC, 2024-11-04:
We do not have existing generic animal tag L22s which would be appropriate linkages, therefore this would require two new terms, something along the lines of: 
Sea Mammal Research Unit (SMRU) Satellite Relay Data Logger (SRDL)
Wildlife Computers TDR10 time depth recorder tags

The TDR10 identifier will need confirming upon creation, but it seems this is their primary if not sole model line. 

We do have existing terms on which these new terms could be based, e.g.: 
TOOL1157	Vemco V9 Coded Transmitter Tag
TOOL1523	Sea Mammal Research Unit Fastloc GSM {GSMDL} GPS phone tag
TOOL1740	Wildlife Computers MiniPAT pop-up archival transmitting tag

gwemon, 2024-11-15:
I think this is perfect! Thanks a lot @JordanAthertonNOC

jakjonbodc, 2024-12-04:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2125/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2126/

jakjonbodc, 2024-12-04:
New terms created and are live on NVS.