nvs-vocabs/L22 #289: NTR: NTR:Thies Precipitation Transmitter 5.4032.

susannetamm, 2025-01-24:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Thies Clima
[This is the name of the manufacturer or institution which made the instrument. If there is a manufacturer but you have not been able to confirm this information or there is no official manufacturer, input 'Inapplicable'. If you are creating a generic instrument (e.g. a generic temperature sensor, then input 'Unspecified'.]

## Precipitation Transmitter
[The manufacturer's instrument model range (e.g. WindObserver). If there is no official model range then input 'n/a'.]

## 5.4032.35.
[Name of the instrument model. If there is no official model name, input 'n/a'.]

## Model version
[Name of the sub-model (e.g. 'D', 'v3.2'). If there is no official version/sub-model then input 'n/a'.]

## Model alternative name (if applicable)
[Alternative name for the model. E.g. Glider Payload CTD alternative name is GPCTD]

## Modification (if applicable)
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plankton net that has been modified since its first description.]

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
[Please indicate what type of instrument it is (e.g. global positioning system, fluorometer, etc.)]

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactured and built according to a design in a paper.]

## Short name
[Shortened name of the instrument. This is usually a shortened version of name limited to make, model and version. It can also be an alternative name.]

## https://www.thiesclima.com/db/dnl/5.4032.35.007-011_5.4032.45.008-009_Niederschlagsgeber_eng.pdf
[Please enter a link to the documentation describing the instrument. This could be a journal reference, manufacturer data sheet, web site, manual etc.]

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

JordanAthertonNOC, 2025-02-12:
Hi @susannetamm 

Thank you for your requests, we will work through these and get back to you as soon as we have created new terms, or if we require further information for any of these.

JordanAthertonNOC, 2025-08-08:
This term was created back in March, apologies it seems this was not posted to this ticket.
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2140/

closed 2025-08-08 by JordanAthertonNOC