nvs-vocabs/L22 #295: NTR: NTR:T40 Temperature sensor

susannetamm, 2025-01-27:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Sea & Sun Technology GmbH
[This is the name of the manufacturer or institution which made the instrument. If there is a manufacturer but you have not been able to confirm this information or there is no official manufacturer, input 'Inapplicable'. If you are creating a generic instrument (e.g. a generic temperature sensor, then input 'Unspecified'.]

## water temperature sensor
[The manufacturer's instrument model range (e.g. WindObserver). If there is no official model range then input 'n/a'.]

## Device model
[Name of the instrument model. If there is no official model name, input 'n/a'.]

## Model version
[Name of the sub-model (e.g. 'D', 'v3.2'). If there is no official version/sub-model then input 'n/a'.]

## Model alternative name (if applicable)
[Alternative name for the model. E.g. Glider Payload CTD alternative name is GPCTD]

## Modification (if applicable)
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plankton net that has been modified since its first description.]

## The temperature sensor T40 is a precise instrument designed for long term measurements under rough conditions. The housing is made of
hard copper and can be used up to 2000m water depth. The Pt100 temperature sensing element is mounted in the bottom end cap , that
allows easy cleaning with high pressure blaster. [Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.

## Instrument type
[Please indicate what type of instrument it is (e.g. global positioning system, fluorometer, etc.)]

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactured and built according to a design in a paper.]

## T40 
[Shortened name of the instrument. This is usually a shortened version of name limited to make, model and version. It can also be an alternative name.]

## Documentation
[Please enter a link to the documentation describing the instrument. This could be a journal reference, manufacturer data sheet, web site, manual etc.]

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

jakjonbodc, 2025-06-13:
Hi susannetamm,

Having looked at your request for Sea & Sun Technology GmbH T40 water temperature sensor I have been unable to find any documents/pages for the T40. The only temperature sensors I have found produced by Sea & Sun Technology GmbH are found on the following link - https://www.sea-sun-tech.com/product-tag/temperature-sensors/. Are you able to please send any further information for the T40, or confirm if your request refers to the SST PT100 or NTC found in the link included?

Many thanks,
Jak

susannetamm, 2025-07-17:
Hi @jakjonbodc ,
I'm also very lost in my colleagues documentation but found the attached one. 
It's propably pretty old and therfore can't be found anymore.

[T 40 Temperature sensor manual - Version 2.pdf](https://github.com/user-attachments/files/21300175/T.40.Temperature.sensor.manual.-.Version.2.pdf)

Best, Susanne

jakjonbodc, 2025-07-17:
Hi @susannetamm 

Thanks for sending this over. I will look over this and get on with creating the new term shortly.

Thanks
Jak

jakjonbodc, 2025-08-08:
Hi @susannetamm 

Term is now live on NVS - https://vocab.nerc.ac.uk/collection/L22/current/TOOL2203/.

Please do get in touch if there is anything else that we can help with.

Thanks
Jak