nvs-vocabs/L22 #296: NTR: Hydroptic Underwater Vision Profiler 6

VBancel, 2025-02-03:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Definition
The Underwater Vision Profiler version 6 or UVP6 is designed to study particles and zooplankton simultaneously and to quantify them in a known volume of water. The UVP counts and sizes all objects > 60 µm and records images of aggregates and plankton > 620 µm. The UVP6 acquires only in-focus images in a volume of water delimited by a light beam issued from a red laser diode in ~100 µs flashes. The collimated light beam illuminates a sampling volume of ~0.6 litre per image and the pixel size is 73µm. The images can be acquired at up to 20Hz. The UVP6 is a very versatile system weighting 1.6Kg in water and consuming very low power. It has been utilised in long term mooring, cabled observatories, CTD-rosette frames, gliders, AUVs, profiling floats and drifting lines. When interfaced with conventional CTD measuring devices, the total abundance of particles can be displayed in real time, integrated with the CTD data. The system has a typical depth range of 6000 m.

## Device make
Hydroptic

## Device range
'n/a'

## Device model
Underwater Vision Profiler version 6

## Model version
'n/a'

## Model alternative name (if applicable)
'n/a'

## Modification (if applicable)
'n/a'

## Qualification (if applicable)
'/n/a'

## Instrument type
Particle and plankton counter using imagery.

## Does the tool describe a series of instruments? 
series

## Reference (if applicable)
https://aslopubs.onlinelibrary.wiley.com/doi/full/10.1002/lom3.10475

## Short name
UVP6

## Documentation
https://aslopubs.onlinelibrary.wiley.com/doi/full/10.1002/lom3.10475

## Mapping to external terminologies (optional)
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1578/
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2107/

## ORCID (optional)
https://orcid.org/0000-0001-8172-5473

danibodc, 2025-02-03:
Hi @VBancel we have a couple of existing L22 terms for the Hydroptic UVP 6:

Standard model: http://vocab.nerc.ac.uk/collection/L22/current/TOOL1578
High frequency model: http://vocab.nerc.ac.uk/collection/L22/current/TOOL2107

Please confirm if one of these would be suitable, many thanks.

VBancel, 2025-02-03:
Hi @danibodc , as discussed via email with Roseanna Wright (vocab.services@bodc.ac.uk is also in CC of the email), we want to create a vocabulary to describe any UVP6 instrument as the parent of the already existing terms, Standard model and HF model.

danibodc, 2025-02-03:
Ah thank you @VBancel!

danibodc, 2025-03-24:
Hi @VBancel this term has now been loaded and will be live on the NVS after the overnight refresh tomorrow morning.

TOOL2141	Hydroptic Underwater Vision Profiler 6 {UVP6} imaging sensor series

danibodc, 2025-03-26:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2141/

closed 2025-03-26 by danibodc

VBancel, 2025-04-17:
Hello @danibodc,

As mentioned in a previous email to Roseanna, would it be possible to map parent/child relationships between vocab terms so that we have the following structure?
UVP5 (https://github.com/nvs-vocabs/L22/issues/301)
parent of:
     --> UVP5 HD (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2110/)
     --> UVP5 StandarD (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2109/)
           parent of:
             --> UVP5 Shallow (https://vocab.nerc.ac.uk/collection/L22/current/TOOL1650/)
             --> UVP5 Deep (https://vocab.nerc.ac.uk/collection/L22/current/TOOL1577/)
             --> UVP5 Zoomed (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2106/)

And for this one : 
UVP6 (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2141/)
parent of : 
         ---> UVP6 LP  http://vocab.nerc.ac.uk/collection/L22/current/TOOL1578/
and   ---> UVP6HF http://vocab.nerc.ac.uk/collection/L22/current/TOOL2107/

Thank you for your feedback!

danibodc, 2025-04-25:
Thanks @VBancel I have loaded the mappings for the UVP6 terms, they will be live on the NVS from tomorrow morning. Roseanna is still working on https://github.com/nvs-vocabs/L22/issues/301 so she has said she will load those mappings once complete.

Many thanks!