nvs-vocabs/L22 #297: NTR: Hemisphere GNSS V200 GPS Compass

JeffSamson, 2025-02-11:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Hemisphere

## Device range
GNSS Smart Antenna

## Device model
Vector V200

## Model version
V200 and V200s

## Model alternative name (if applicable)
Vector V200 GNSS Smart Antenna

## Modification (if applicable)
None

## Qualification (if applicable)
N/A

## Instrument type
Global Navigation Satellite System (GNSS) Recevier

## Does the tool describe a series of instruments? 
There are variations in models: V200 and V200s

## Reference (if applicable)
https://www.subseatechnologies.com/media/files/page/98256cd1/v200-pn.pdf

## Short name
Vector V200 GNSS Smart Antenna

## Documentation
https://www.subseatechnologies.com/media/files/page/98256cd1/v200-pn.pdf

## Mapping to external terminologies (optional)
This model ins the next generation of an existing L22: https://vocab.nerc.ac.uk/collection/L22/current/TOOL1514/

## ORCID (optional)
[Enter your ORCID if desired]

danibodc, 2025-05-08:
This term has now been loaded and will be live on the NVS from tomorrow morning:
TOOL2162	Hemisphere GNSS Vector V200 GPS compass series

danibodc, 2025-05-09:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2162/

closed 2025-05-09 by danibodc