nvs-vocabs/L22 #298: NTR: Garmin GPSMAP 78 series

JeffSamson, 2025-02-11:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Garmin

## Device range
GPSMAP

## Device model
78 series

## Model version
N/A

## Model alternative name (if applicable)
Handheld GPS

## Modification (if applicable)
N/A

## Qualification (if applicable)
Handheld GPS

## Instrument type
global positioning system

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
N/A

## Short name
GPSMAP

## Documentation
https://static.garmincdn.com/pumac/GPSMAP_78_OM_EN.pdf

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
N/A

danibodc, 2025-05-08:
This term has now been loaded and will be live on the NVS from tomorrow morning:
TOOL2161	Garmin GPSMAP 78 handheld marine Global Positioning System receiver series

danibodc, 2025-05-09:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2161/