nvs-vocabs/L22 #299: NTR: Sea-Bird Deep SeapHOx V2

ONC-KWright, 2025-02-21:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Sea-Bird Scientific

## Device range
SeapHOx

## Device model
Deep SeapHOx

## Model version
V2

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
In House Modification 'adapting the MBARI/SIO Deep-Sea DuraFET technology'

## Qualification (if applicable)
Deep

## Instrument type
Integrated Conductivity Temperature Depth pH O2 instrument

## Does the tool describe a series of instruments? 
Series of Systems

## Reference (if applicable)
N/A

## Short name
Deep SeapHOx V2

## Documentation
https://www.seabird.com/deep-seaphox-v2-ocean-ct-d-ph-do-sensor/product?id=60762467726

## Mapping to external terminologies (optional)
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1895/

## ORCID (optional)
0009-0001-5053-2959

ONC-KWright, 2025-04-15:
Hello @roswri ,

The two instruments use different ISFET sensors. From [Shallow vs. Deep SeapHOx V2](https://www.comm-tec.com/Docs/AN/sbe/Technical_Note_ShallowVsDeepSeapHOx.pdf):
> While both harness the same underlying technology, the shallow and deep versions have a different ISFET and reference electrode construction. 

Comparing the [Shallow SeapHOx manual](https://www.seabird.com/asset-get.download.jsa?id=69833855331):
> The primary sensor element of the SeaFET™ is the ISFET, a solid-state sensor that
senses pH in marine environments. The ISFET has two reference electrodes: an internal
reference and an external reference, that give separate reference potentials to the ISFET
and show separate pH values (pH Internal and pH External). After the corrections for
temperature and salinity are applied, the values from the internal and external are similar,
and let the user verify the validity of the sensor's measurements.
**Internal reference**
The internal reference electrode inside the DuraFET® is immersed in a bath of saturated
potassium chloride (KCl) gel and is physically separated frm the environment. The KCl
gel exposes the Ag/AgCl internal electrode to a relatively constant chloride concentration.
The sensor can therefore measure pH regardless of environmental salinity. If accurate
salinity and temperature data are not available, the internal cell is generally more
accurate.
**External reference**
The external reference electrode has a Ag/AgCl reference electrode in direct contact with
seawater. The potential of this electrode varies with pH and chloride concentration, so
unless the chloride concentration is known, the external reference is not stable. To
correct this, salinity can act as an approximation of chloride concentration. If accurate
salinity data is available, it can be applied to the pH external data and significantly reduce
measurement errors, and give the most accurate and stable pH data

with the [Deep SeapHOx manual](https://www.seabird.com/asset-get.download.jsa?id=69833853782):
> The primary sensor element of the Deep SeapHOx V2 is a special deep-packaged
ISFET, a solid-state sensor that senses pH in marine environments.
**External reference**
The external reference electrode has a Ag/AgCl reference electrode in direct contact with
seawater. The potential of this electrode varies with pH and chloride concentration, so
unless the chloride concentration is known, the external reference is not stable. To
correct this, salinity is used as an approximation of chloride concentration. The salinity
data that is calculated by the SBE-37-SMP-ODO is applied to the pH external data and
significantly reduces measurement errors, and gives the most accurate and stable pH
data
at section 4.1 in both we can see that the Shallow SeapHOx has an additional internal reference electrode that isn't on the Deep SeapHOx.

Additionally, in [Application Note 99 - Calculating pH from ISFET pH Sensors](https://www.seabird.com/asset-get.download.jsa?id=69833850609) we can again see a calculation for a pH Internal that isn't applicable to the Deep SeapHOx.

Please let me know if you need any more information.

Cheers,
Keegan

roswri, 2025-05-01:
Hi @ONC-KWright ,

Many thanks for clarifying, the new term is now live on the NVS: https://vocab.nerc.ac.uk/collection/L22/current/TOOL2153/

closed 2025-05-01 by roswri