nvs-vocabs/L22 #302: NTR: Aanderaa Optode 4797 pCO2

stephgolob, 2025-03-13:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Aanderaa 

## Device range
N/A

## Device model
4797

## Model version
6.0.11.0

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Carbon dioxide sensor

## Does the tool describe a series of instruments? 
Individual

## Reference (if applicable)
Atamanchuk, D., Tengberg, A., Thomas, P. J., Hovdenes, J., Apostolidis, A., Huber, C., & Hall, P. O. (2014). Performance of a lifetime‐based optode for measuring partial pressure of carbon dioxide in natural waters. Limnology and Oceanography: Methods, 12(2), 63-73.

## Short name
N/A

## Documentation
Documentation is available in an internal documentation service - is there somewhere I could share the file?

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0001-7662-5996

JordanAthertonNOC, 2025-03-14:
Hi Steph, thanks for this request. In this instance you can send the documentation to myself (jordan.atherton@noc.ac.uk) but in future feel free to email the whole team (vocab.services@bodc.ac.uk) if you have supporting documentation.

For public documents you can attach them to the ticket 😃

ahpanton, 2025-05-01:
Hi @stephgolob - I have nearly finished the L22 for this instrument. Thank you for providing the reference and additional documentation. One quick question - would it be fair to describe this sensor as a 'prototype' in the description as I don't see it listed on the Aanderaa website at all?

ahpanton, 2025-05-09:
Thank you for confirming that. I have created this L22 for you and the term will be live on the NVS from tomorrow:

TOOL2163 - Aanderaa 4797 pCO2 optode prototype

ahpanton, 2025-05-13:
Now live on NVS:

[TOOL2163](http://vocab.nerc.ac.uk/collection/L22/current/TOOL2163/) - Aanderaa 4797 pCO2 optode prototype

closed 2025-05-13 by ahpanton