nvs-vocabs/L22 #303: NTR: RST Thermistor Strings

stephgolob, 2025-03-28:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
RST Instruments Ltd.

## Device range
N/A

## Device model
2252 Ohm Thermistors

## Model version
N/A

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
RST Thermistor Strings

## Does the tool describe a series of instruments? 
System

## Reference (if applicable)
N/A

## Short name
N/A

## Documentation
https://rstinstruments.com/product/thermistor-strings/

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0001-7662-5996

stephgolob, 2025-03-28:
My team and I are unclear on whether this device constitutes a series of instruments or a system. Is there somewhere on the NVS website that provides an explicit definition?

JordanAthertonNOC, 2025-04-03:
Hi @stephgolob, we don't have anything published that explicitly describes the differences between instruments, systems, and a series of instruments unfortunately, but it is something we may now look to include in our Guides and Resources repo eventually. 

Thankfully it doesn't impact this ticket though, we'll have a sit down and evaluate what you've provided and come to a decision. Typically systems are in the navigation space, where several data sources are combined to provide a combined output for example. At a glance your request is looking more like a series, but we'll take a deeper look.

ahpanton, 2025-05-14:
Hi @stephgolob please could you let me know some more about how these thermistor strings are being used/deployed? I have been in touch with RST Instruments for further details but it would be helpful to hear about some real-world deployments as well as the manufacturer specifications. If the details are confidential in any way then please feel free to email me: anopan@noc.ac.uk.
Thank you very much
Anouska

ahpanton, 2025-05-15:
Also please can I check the resistance rating? The information above in 'Device Model' states 2252 Ohm thermistors (which I'm assuming is at 25 degrees C) yet the RST Instruments website and specification sheets state 3k ohms at 25 degrees C. I can find other manufacturers that provide thermistors at both of these resistance ratings so I would just like to clarify.
Many thanks
Anouska

ahpanton, 2025-07-03:
Term now live on the NVS:

[TOOL2182 - RST Instruments thermistor strings series](http://vocab.nerc.ac.uk/collection/L22/current/TOOL2182/)