nvs-vocabs/L22 #304: NTR: Wally Benthic Crawler

bhrtd, 2025-04-07:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
METAS AS

## Device range
n/a

## Device model
PCI-160

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
'Wally I' was modified significantly to become 'Wally III' as its new retrofit version.

## Qualification (if applicable)
n/a

## Instrument type
METAS Crawler Interface Container

## Does the tool describe a series of instruments? 
Networked sensor communication tools (compass sensors, motor controllers, Ethernet switches, serial servers, relay servers, DAQ servers, optional video encoders, stereo cameras, and pan/tilt units for monitoring).

## Reference (if applicable)
n/a

## Short name
Wally III

## Documentation
* Manufacturer website and datasheet: https://www.offshore-technology.com/contractors/auvs-rovs-diving-and-cameras/metas-as/
* The Metas PSI-160 Interface Container, produced by METAS and supplied by Jacobs University, is a modular unit designed for seamless sensor communication. It plays a crucial role in various Metas products by enabling efficient data exchange between sensors and external systems. The METAS Benthic Crawler, also known as Wally, is an internet-controlled benthic crawler operated by researchers at Jacobs University in Bremen, Germany. It is built from durable carbon fiber and can withstand pressures from 150 to 400 bar. Wally’s modular design supports integrating various platforms, including X-Frame, X-Sat, MUST, and ROSSIA. It features two main sections—a power section and an interface section—positioned on opposite container lids, with ample space for additional payload. Its versatile architecture supports a Master-Slave configuration, and the smart communication interface can connect to an external mission planner module for enhanced operational efficiency. Wally is equipped with a range of sensors, including a compass sensor (Magnetic Compass Heading, Pitch, Roll, Internal Compass Temperature) embedded within a TCM-XB 12397 V2.8, and motor-related sensors (Right Motor Current, Left Motor Current) integrated into an embedded motor. Wally has been used to explore methane gas hydrate fields in Barkley Canyon, off the coast of Vancouver Island, British Columbia.

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
n/a

[5212-1120-B+Manual+for+PCI+160+Interface+Container+for+Project+3332+Craw....pdf](https://github.com/user-attachments/files/19637922/5212-1120-B%2BManual%2Bfor%2BPCI%2B160%2BInterface%2BContainer%2Bfor%2BProject%2B3332%2BCraw.pdf)

bhrtd, 2025-04-23:
Hi, I just wanted to follow up on the status of this request. Thank you!

JordanAthertonNOC, 2025-04-24:
Hi @bhrtd we have assigned this now and will get a term for you ASAP, apologies for the delay.

roswri, 2025-04-24:
Hi @bhrtd ,

If I understand correctly the Wally III is a benthic crawler system that can carry sensors and cameras, and the METAS PCI-160 is the module used in the Wally III crawler to enable remote operation of the crawler and facilitate communication with the sensors/data acquisition. The METAS PCI-160 Crawler Interface Container could also be used in other products, not only the WALLY III.

We can create a term for the METAS PCI-160 Crawler Interface Container (without mention of Wally III) - could you confirm this meets your requirements?

Depending on your metadata requirements, you might also want to consider creating a B76 term for the Wally III platform, and additional L22 terms for the camera/sensors mounted on the Wally III system.

bhrtd, 2025-04-24:
Hi @roswri,
Thanks for your comment! Yes, that aligns perfectly. The METAS PCI-160 is more of a modular interface unit, while Wally is a self-contained autonomous underwater crawler. Your proposed approach makes sense.
Thank you!

roswri, 2025-05-15:
Hi,
New term created and is now live on the NVS: https://vocab.nerc.ac.uk/collection/L22/current/TOOL2155/

closed 2025-05-15 by roswri

bhrtd, 2025-05-15:
Hi @roswri,
Thank you so much for the update!