nvs-vocabs/L22 #308: NTR:

ONC-KWright, 2025-04-30:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
In attached .xlsx file

## Device range
In attached .xlsx file

## Device model
In attached .xlsx file

## Model version
In attached .xlsx file

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Global Positioning System

## Does the tool describe a series of instruments? 
N/A

## Reference (if applicable)
N/A

## Short name
In attached .xlsx file

## Documentation
In attached .xlsx file

## Mapping to external terminologies (optional)
No

## ORCID (optional)
[0009-0001-5053-2959](https://orcid.org/0009-0001-5053-2959)

[L22_GPS_Tablet_Request.xlsx](https://github.com/user-attachments/files/19981919/L22_GPS_Tablet_Request.xlsx)

hanrev, 2025-05-29:
Hi Keegan, for this GPS tablet term would you be OK with this being kept as a more generic term e.g. something loosely along the lines of:

'Unspecified tablet GPS receiver'

as this isn't a specific oceanographic instrument and all tablet devices will be using the same GPS/GLONASS satellite system to receive the positional information. Are there any details associated (e.g. accuracies/precisions) with the GPS receiver in this specific tablet make and model that would make specifying the manufacturer etc. in the L22 a must?

hanrev, 2025-06-09:
This term has now been created and will be published online tomorrow as follows: Unspecified tablet computer Global Navigation Satellite System (GNSS) receiver (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2167/)

closed 2025-06-10 by hanrev