nvs-vocabs/L22 #309: NTR: TA Instruments Q600

whshannon, 2025-06-05:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
TA Instruments

## Device range
Simultaneous Thermal Analyzer

## Device model
Q600

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
 simultaneous thermal analyzer; provides simultaneous measurement of weight change (TGA) and true differential heat flow (DSC) 

## Does the tool describe a series of instruments? 
No

## Reference (if applicable)
n/a

## Short name
TA Instruments Q600

## Documentation
Manual from TA instruments: https://www.tainstruments.com/wp-content/uploads/sdt.pdf

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-9081-1004

hanrev, 2025-07-15:
Good afternoon @whshannon, 
Would you be able to provide some details on what this instrument was used to analyse?

hanrev, 2025-08-01:
Good afternoon,
thank you for your patience with this! 

This term has now been created and will be published online tomorrow as follows: TA Instruments SDT Q600 simultaneous thermal analyser (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2193/)