nvs-vocabs/L22 #320: NTR:SilCam

DanaMaKing, 2025-07-09:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Sintef Ocean AS

## Device range
'n/a'

## Device model
In situ particle imaging system

## Model version
 'n/a'

## Model alternative name (if applicable)
'n/a'

## Modification (if applicable)
'n/a'

## Qualification (if applicable)
'n/a'

## Instrument type
In situ particle imaging system

## Does the tool describe a series of instruments? 
'system'

## Reference (if applicable)
https://www.sciencedirect.com/science/article/pii/S0025326X16309833

## Short name
SilCam

## Documentation
https://www.sciencedirect.com/science/article/pii/S0025326X16309833

## Mapping to external terminologies (optional)
'n/a'

## ORCID (optional)
'n/a'

DanaMaKing, 2025-08-15:
Hey @danibodc ,
Thank you very much for creating the concept. However the SilCam doesn't use holography. Is it possible to remove that ?

DanaMaKing, 2025-08-15:
Hey @danibodc , 
Perfect, thank you so much :)

danibodc, 2025-08-15:
@DanaMaKing the edits have been made, it is now queued for overnight publication so won't be live on the NVS until tomorrow morning.