nvs-vocabs/L22 #328: NTR: Metrohm 800 Dosino

whshannon, 2025-07-25:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Metrohm

## Device range
n/a

## Device model
800

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
Dosing device suitable for titrations and liquid handling tasks

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
https://www.metrohm.com/en_us/products/2/8000/28000010.html
https://www.metrohm.com/content/dam/metrohm/shared/documents/manuals/88/88008002EN.pdf

## Short name
Metrohm 800

## Documentation
https://www.metrohm.com/en_us/products/2/8000/28000010.html
https://www.metrohm.com/content/dam/metrohm/shared/documents/manuals/88/88008002EN.pdf

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

hanrev, 2025-09-19:
Hi @jakjonbodc, the Autolab PG101 Potentiostat has a separate ticket which has been assigned to me (#330) - and the Metrohm 858 sample processor (#327) is one that I created a few weeks ago. After having seen these separate requests I agree it may be more suitable to create this as one system L22 but I think this needs further discussion. 
@whshannon, are the Metrohm 843 Peristaltic Pumps (#329) also part of this same system?

whshannon, 2025-09-19:
Yes, they were! Here's the metadata page describing data collection: https://www.bco-dmo.org/dataset/969915 (see "Methods & Sampling")