nvs-vocabs/L22 #334: NTR: KduPRO

Carlos-Rodero, 2025-07-31:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Institute of Marine Sciences - CSIC

## Device range
'n/a'

## Device model
KduPRO

## Model version
Multispectral

## Model alternative name (if applicable)
KduPRO-Multi

## Modification (if applicable)

## Qualification (if applicable)
Description: The KduPRO is a cost-affordable and DIY (Do-It-Yourself) moored system evolved from the KdUINO (Bardaji et al, 2016). It is based on a modular system of light sensors, independent of each other, measuring the irradiance at different depths. The depth of each module can be modified according to the requirements of the project or the environment, offering the user a custom array of sensors. 
The KduPRO multispectral measures the light intensity at wavelengths of 415, 445, 480, 515, 555, 590, 630, 680 and 910 nm and PAR (Photosynthetically Active Radiation) band at several depths in the water column to estimate the diffuse attenuation coefficient, Kd.

'cost-affordable', 'DIY', 'moored system', 'modular system of light sensors',  

## Instrument type
Underwater light sensor

## Does the tool describe a series of instruments? 
Series of instruments

## Reference (if applicable)
(Rodero et al, 2022)

## Short name
KduPRO-Multi

## Documentation
https://zenodo.org/records/5721155, https://doi.org/10.3389/fmars.2022.1004159

## Mapping to external terminologies (optional)

## ORCID (optional)
https://orcid.org/0000-0002-9331-6519

Carlos-Rodero, 2025-08-04:
> Hi @Carlos-Rodero <https://github.com/Carlos-Rodero>
>
> Technically for this request we could create a term for the APDS-9960 RBG
> light sensor as ultimately this appears to be the instrument generating the
> data, however I understand your request is slightly different. I can
> therefore create a new term for either:
>
>    1. A single unit of the KduPRO (e.g. something like Institute of
>    Marine Sciences KduPRO radiometer)
>    2. A series of units (e.g. something like Institute of Marine Sciences
>    KduPRO radiometer system)
>
> What is your preference for your use-case? Both would specifically
> reference the APDS-9960 RBG light sensor module in the term description.
>
> Additionally, unless you are opposed, I will likely omit CSIC from the
> term name unless you specifically require this - for similar instruments we
> would not include the parent funding council, e.g. the following NOC term
> does not include our research council (NERC).
> https://vocab.nerc.ac.uk/collection/L22/current/TOOL1897/
>
> However it would have a linkage the vocabulary collection for
> organisations (B75) which does include CSIC in the title for ICM.

Hi Jordan,

The sensor is specifically the AS7341, but I prefer to codify as a whole
with the name of KduPRO, and this model specifically as KduPRO-multi. But
yes, the sensor has been defined in the term description.
We have previously worked with other KduPRO (model KduPRO-RGB) devices
using various light sensors, such as the TCS34725. I believe the same
approach will be taken in the future, where we continue to improve for
other light sensors.
For these reasons, I think the instrument itself is the KduPRO.

A buoy of this instrument has a N number of KduPRO (for instance, 5 sensors
measuring at the same time, to estimate the Downwelling Diffuse Attenuation
Coefficient of the X spectral band. So maybe it is close to the
definition: A series of units (e.g. something like Institute of Marine
Sciences KduPRO radiometer system)

*Additionally, unless you are opposed, I will likely omit CSIC from the
term name unless you specifically require this - for similar instruments we
would not include the parent funding council, e.g. the following NOC term
does not include our research council
(NERC). https://vocab.nerc.ac.uk/collection/L22/current/TOOL1897/
<https://vocab.nerc.ac.uk/collection/L22/current/TOOL1897/>*
Thats fine!

Let me know if its OK for you, and thanks for your help,

Carlos

Missatge de Jordan Atherton ***@***.***> del dia dl., 4 d’ag.
2025 a les 17:21:

JordanAthertonNOC, 2025-08-05:
Thanks Carlos

Could you clarify the light sensor component usage, is this interchangeable or are there two distinct models you have developed? 

One as the KduPRO utilising the TCS34725/APDS-9960, and a second as the KduPRO-multi utilising the AS7341, or does the -multi simply denote a series of units?

JordanAthertonNOC, 2025-08-05:
I'm happy to create as many terms as required to meet your requirements, for example I could create the following to cover differing use cases, so the singular units can also be characterised. I can also link the two groupings with broad - narrow (and inverse) mappings.

Institute of Marine Sciences KduPRO-RGB radiometer system
Institute of Marine Sciences KduPRO-RGB radiometer

Institute of Marine Sciences KduPRO-Multi radiometer system
Institute of Marine Sciences KduPRO-Multi radiometer

JordanAthertonNOC, 2025-08-05:
Do you have the wavelengths for the RGB model?

Carlos-Rodero, 2025-08-05:
Perfect!

The wavelengths not exactly, but they are approximately in these nm:
[image: image.png]

Missatge de Jordan Atherton ***@***.***> del dia dt., 5 d’ag.
2025 a les 12:36:

JordanAthertonNOC, 2025-08-05:
Hi Carlos, the four new terms as below:

A new term has been created and will be published online tomorrow as follows: Institute of Marine Sciences KduPRO-RGB radiometer system (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2196/)
A new term has been created and will be published online tomorrow as follows: Institute of Marine Sciences KduPRO-RGB radiometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2197/)

A new term has been created and will be published online tomorrow as follows: Institute of Marine Sciences KduPRO-Multi radiometer system (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2198/)
A new term has been created and will be published online tomorrow as follows: Institute of Marine Sciences KduPRO-Multi radiometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL2199/)

Carlos-Rodero, 2025-08-05:
Thank you so much Jordan! 
Thanks for your time and effort :)

JordanAthertonNOC, 2025-08-06:
In your NetCDF attributes if it's helpful:

For `sensor_manufacturer` you could consider using the following manufacturer vocabulary:
https://vocab.nerc.ac.uk/collection/L35/current/MAN0349/

And for `sensor_reference` you could use the following persistent links (for RGB and Multi respectively) to the light sensor components, as the links to the commercial website may become unavailable in the future, whereas the below links are intended to be persisent:
https://www.bodc.ac.uk/data/documents/nodb/pdf/TCS34725.pdf
https://www.bodc.ac.uk/data/documents/nodb/pdf/AS7341.pdf