nvs-vocabs/L22 #340: NTR: Seaglider SGX

rhudak2020, 2025-08-07:
### Device make

University of Washington

### Device range (if applicable)

Seaglider

### Device model

SGX

### Model version (if applicable)

n/a

### Model alternative name (if applicable)

n/a

### Modification (if applicable)

n/a

### Qualification (if applicable)

n/a

### Instrument type

glider

### Instrument category?

System

### Documentation

https://iop.apl.washington.edu/iopsg/SGX_documentation_2024-v1.0.pdf

JordanAthertonNOC, 2025-08-08:
Hi @rhudak2020 

This exists in our platform vocabulary here:
https://vocab.nerc.ac.uk/collection/B76/current/B7600034/

closed 2025-08-08 by JordanAthertonNOC

rhudak2020, 2025-08-08:
Thank you, I'll have to expand into different vocabularies haha I've been concentrating on L22 and L05.

https://vocab.nerc.ac.uk/collection/B76/current/B7600034/